Moyster/nougat_device_meizu_m2note #1: Video

7KiLL, 2017-02-05:
Hello again, Moyster!
New ROM is awesome! Fast, clean and pretty. But it also has some bugs. Video playback. 
First:
As I've been writing to you before, it's Twitch.TV stream app. Now it just gives black screen w\o sound.
Second: 
Youtube. It gives an error "No network" (or smth like that, im on russian locale) for every video. 

I have some problems with logcat, so please, try those apps by youself, I'm pretty sure it gives to you more information than me. Thanks!

Moyster, 2017-02-05:
Yeah, don't worry I might have did last minutes changed and it broke omx :D 

I'll fix that tomorrow or in a couple days ;)

Moyster, 2017-02-06:
https://github.com/Moyster/nougat_device_meizu_m2note/commit/66ac3746c8987bf8d48cee602a91f81c9cc8c7b6 should fix it ;)
I'll post a new build tomorrow evening, but right now, Go Falcons!

Moyster, 2017-02-07:
It's fixed, also it's worth mentionning if you only have 240<->360p in youtube, you need to clean youtube's data and/or reinstall it.

closed 2017-02-07 by Moyster

7KiLL, 2017-02-07:
Yea, thanks, dude. I know this cuz I've answered for this question on XDA lol