Moyster/nougat_device_meizu_m2note #4: Update audio_policy.conf; Update audio_effects.conf; Update audio_device.xml

teik-off, 2017-02-06:
https://github.com/dmitrywxz/LineageOS_device_meizu_m2note/commits/los-14.1

Moyster, 2017-02-06:
Makes no sense, doing changes just to make changes is pointless.

"AUDIO_DEVICE_IN_FM"
Do you really think we have an FM INPUT on a device without FM radio ? I even removed FMradio driver from the kernel =/

closed 2017-02-06 by Moyster

teik-off, 2017-02-06:
Only this is tested and really works. The sound is louder and cleaner!

teik-off, 2017-02-06:
He agrees, has gone too far in FM.
But, the sound becomes better if to make so:
- sampling_rates 44100
+ sampling_rates 44100|48000

teik-off, 2017-02-06:
He agrees, has gone too far in FM.
But, the sound becomes better if to make so:

   - sampling_rates 44100

   - sampling_rates 44100|48000

Moyster, 2017-02-07:
make it a pull request please, thanks :)