Moyster/nougat_device_meizu_m2note #5: Snapchat Record

Carron8, 2017-02-07:
I can't record any video with the app "Snapchat" the app give me an error when I try record any vidieo, pls try fix this bugg. Thanks for all work

Moyster, 2017-02-07:
Hmmm well if someone is interested in Snapchat videos, this is what you should try :  
https://github.com/Moyster/nougat_device_meizu_m2note/blob/los-14.1/configs/media/media_profiles.xml  
This file contains the video profiles for various things, including recording videos, as of now, I think "1080p" recording works, but the profiles for 720p doesn't.  
Why ? Because the profiles listed as 720p and timelapse720p have wrong resolutions. currently we have 640x480 for 720p instead of 720x480.  
I don't know if snapchat uses the VGA or Qcif profiles, but that's definitively something to investiguate ;)  

The easiest way is to try with "Snap" to record videos, test the various profiles, see which ones work.
720p should not work, but editing media_profiles.xml and replacing the resolution from 640x480 => 720x480 fixes it.  
notepad, a bit of free time, "adb push" and it could be fixed ;)  

To anyone willing to test and fix the video profiles, thank you for your pull requests ;)

dmitrywxz, 2017-02-07:
I myself have collected on your tree.and it all started. why not start here?I can understand the boast on the fact that no one would have repeated. but why would a tree put?

Carron8, 2017-02-07:
Thx I don't know how i can use adb to fix sorry.

Carron8, 2017-02-08:
I try later to use ADB to chose in my phone thx for all work.

Carron8, 2017-02-09:
I put the media_profiles.xml but record in "Snapchat" don't works thx for all work pls try to fix this problem to do the best rom.

Carron8, 2017-02-13:
I put this file and restar, now insagram video work but Snapchat can't record

closed 2017-02-14 by Carron8