Moyster/nougat_device_meizu_m2note #6: Does not start

dmitrywxz, 2017-02-07:
Can I hear a plausible answer on why not kick off the firmware?

dmitrywxz, 2017-02-07:
I understand the patches were applied, but in the tree they are not.In fact, why do so?And the vendor can is to register the file libstlport.so? well, if he's in the final firmware is present.

dmitrywxz, 2017-02-07:
# Sign the build with private certs
PRODUCT_DEFAULT_DEV_CERTIFICATE := vendor/moyster/keys/releasekey
Where to get?

dmitrywxz, 2017-02-08:
Where are the patches? after all, without them does not work. that's for sure.

Moyster, 2017-02-08:
You're... amazing.... "plausible answer", not a single mark of courtesy, like saying Hi, please, thanks. 
Did we raise the cows together in our youth ? I don't think so.

Next time you have something to ask, start by asking it properly.
Now since you deleted most of your comments but my mailbox kept a trace of it...  
- Libstlport is built from sources
- signing keys are private certificates you can generate on your own.  
  
Now for the "plausible" reason it doesn't boot FOR YOU, here's a simple answer : look at "your" work, good job doing copy/paste, kangs, but not doing the single thing that can help you : learning.

You're also, you and many others disrespectful "romers" why there's no patches in that device tree, y'all just want to fork and build over my work just so you can call it yours, look what was done to cm-12.1.
Here's where are the patches, I started simple, booted Nougat, fixed what I could straight from the vendor tree so we don't have to use patches, example : libui_ext / camera blobs / drm blobs.
Did you say thanks ?? 
After that, I realised nobody had any patches online for Nougat, so I looked at what was done previously... and fixed it and/or coded it myself.

Good luck, Have fun. Now is the time for you to show you can pull some work on your own, you won't get any help from me with that behavior.

Have a good day.

closed 2017-02-08 by Moyster

dmitrywxz, 2017-02-08:
lol

Moyster, 2017-02-08:
Nice proof of maturity, deleting your entire comments, github, etc... LMAO.

[image]

:100: :100:  :1st_place_medal: 

[image]
I said no, you're pissed, you send me this... but it can't be deleted :) "you really did everything without me" ? LMAO. At least you're nice enough not to come here and tell me I "forked" your work :+1: which I didn't, or it won't boot. lmao.
[image]

Looking at your "friends", and the dates, who's being rude / harrassing who exactly ?

[image]

That's exactly what I guessed. So, let this comment be clear to all of you and your friends. I don't care about your age, you can be 16 or 33, what comes out of your mouth matters, and in a "dev community", what comes out of your brain, matters. Obviously those two conditions are not satisfied.
If you think what I do is "shit" or "useless" or whatever, DONT USE IT, DONT COME TO ME FOR HELP, DONT BE RUDE, DONT GET UPSET JUST BECAUSE YOU FORGOT YOU'RE A PERFECT RANDOM WITH ABSOLUTE NO PRIVILEGES.

You keep doing this BS, it will have real consequences, boy, you don't know me, you don't harrass me. 

I'M THE DANGER, I'M THE ONE WHO KNOCKS.

Take this advice seriously, Don't -EVER- dare fuck with me. Make it crystal clear.

dmitrywxz, 2017-02-08:
Just do not have to write any nonsense.Your trees and everything, it all just taken from others.And then you claim something?Disable you Internet connection, and you yourself will not do anything without Google.So you don't tell someone something, you have taken.Calm down kid and breathe.

dmitrywxz, 2017-02-08:
Oh, by the way, if you're as well versed in the code, like moving the brain.Then fix the sound. You have him, sad shit.And quiet as a mouse.

Moyster, 2017-02-08:
"Just do not have to write any nonsense.Your trees and everything, it all just taken from others.And then you claim something"  
You're such a joke :') Go ahead, find where I copied and fixed my rootdir files, find a single tree without tinymix built (I don't :D).  
Find someone who reworked libmtk_symbols into a shim lib ???  
Find someone who published a fix for the camera ? Is this copypasta too ?  
Who's got legacy mtk omx working on lineage 14.1 ? And I'm not talking about swapping with broken omx blobs xD.
I don't have any youtube zooming problem, unlike 90% Nougat build (Exynos / MTk and so on....)

You don't know what you're saying, you're salty as fuck, and the only reason you keep arguing is because you think WE ARE EQUAL, obviously you don't understand a single thing about what are device trees... 
Just because you saw whatever looked similar to someone else's work, doesn't mean I copied him.

So... Playing the differences game... Who's building libstlport from sources ? Divis1969 uses prebuilt from CM/Lineage, it's in his device tree.  
This mean : 
- CM/lineage OFFERS PREBUILT FOR THAT  
- I compile them from sources  
Yet there's only ONE PERSON WHO CAN'T FIGURE A WAY XD  

You're full of shit, "Then fix the sound. You have him, sad shit.And quiet as a mouse."
who's the sad shit coming here ALL DAY wishing to find the PATCHES and HELP ? XDDD
And seriously I have no problems with sound, unlike you I don't spend the day listening to music because that's all i have. ROFL.

By the way, I'm still a student, I was at work, it's called an internship, I was deploying and assisting a fibber network deployment, back at the office... Just to find your miserable person commenting here.
Who's the silent mouse ?  
You need what I "have to say", and obviously, 
GO AHEAD AND PESTER THEM?  

In the end it's totally crazy, arguing with someone who can't understand he's wrong. a wrong NOOB.

Moyster, 2017-02-08:
BTW: you have my kernel, my device tree, my vendor tree. But it's still not booting. Tell me I suck and I didn't pull CODE into this ;) ;) ;) ;)

dmitrywxz, 2017-02-08:
You to describe in detail how you pieces all take? or he remember?There's nothing truly yours. I mean from beginning to end.And pieces of glue code and adjust it for themselves, many do not mind.Write the engineering menu on см14 myself from scratch without Google, gihub and other assistance.and I believe that you can do something. well you're just a windbag.

dmitrywxz, 2017-02-08:
At the expense of sound.you may not listen to it.I don't know. but you post the firmware.think about the end users and not just about yourself.But the sound and in truth can be improved. I improved the quality and volume.if you are jealous, envious silence.

dmitrywxz, 2017-02-08:
as we say - bullshit no bags to roll.Is that really to say, and say shit not carry it.

dmitrywxz, 2017-02-08:
Do not be offended and do not think that I am a fool. it is not necessary.I understand perfectly.If something essentially write.I am not evil and not stupid. just me losing my temper.but you be appropriate. don't need to be a jerk.