Moyster/nougat_device_meizu_m2note #11: Whatsapp videos won't play

gloddo, 2017-02-17:
I'm sorry to bother, I don't even consider this a major bug but I thought you'd like to know every single one happens.
When I open a video, sent or received, directly from whatsapp's chat, It won't play, saying that "gallery app has interrupted"

[image]

gloddo, 2017-02-17:
That's a huge screenshot, sorry for that lol

franci22, 2017-02-17:
Try to use another video player, like VLC.
---
Prova ad usare un altro player come VLC :)

gloddo, 2017-02-17:
It worked, but it's just a workaround, I think a fully working stock app would be better, also not to bloat the system with too many apps

Moyster, 2017-02-19:
Hello,

If you can make a logcat of the moment when you open that video and it crashes, it'll be better than a screenshot ;)
I'm wondering tho, is the video crashing gallery app OR only crashing when opening FROM what'sapp ?

gloddo, 2017-02-20:
I tried a couple apps for logcats but I couldn't do it, I'm sorry.. Anyway, I tried opening videos directly from gallery, it prompts me if I want to use default "video player" app and that makes the galley app crash.

gloddo, 2017-02-25:
This is fixed in beta 4. Thank you

closed 2017-02-25 by gloddo