Moyster/nougat_device_meizu_m2note #13: After a while amy video can't be played

gloddo, 2017-02-21:
Auto-playing videos on facebook make the app crash, so does any youtube video. Videos stored on the phone and opened with VLC play without problems, but at least those two apps are not working anymore. They used to work at the beginning. The problem came out after about 4 days.

gloddo, 2017-02-21:
After a reboot it came back to normal, but before that every sound stopped. Ringtone, alarm, also unlock and tap sound. I was kinda scared actually.. I don't want to flash flyme back 😓

Moyster, 2017-02-21:
Hi there,  

What build ? Did you flash any mod (SuperSU / Dolby atmos / any ) ? Did you try flashing after doing a clean wipe as between (beta) builds, there was some changes.
You could also try manually cleaning the datas for for both crashing apps (settings > apps > etc... > storage > clear all data)  
Making logs would be useful also, as I'm not a magician, and there's plenty of tutorials on Xda / Youtube to learn how to make some ;)

Thanks.

gloddo, 2017-02-21:
I came from flyme directly to beta 3, and during first flash I installed SuperSU and Viper+dolby atmos. I'll try my best to give you a log if anything of this happens again. Right now everything is working fine, except for videos that sometimes do some weird green flashes, but that's it. Thank you for your advices and for the amazing effort you are putting into this project.. This phone is still pretty cool, but with this rom it has become amazing ;)

Moyster, 2017-02-21:
Well, I suspect some codecs to be borked/buggy, you could check what video/audio codecs are causing issues, then report it here, I'll do some changes in the coming build to see how it goes, but I know tomax80 (on xda) has issues with some videos using Kodi on m2note, mostly mp4 if I remember correctly.  

But since you have viper+dolby, you might want to do a backup, flash clean and test for a few hours/days if that's not "the cause", I know someone reported "noisy a2dp" when using the dolby mod, so I don't trust "mods" anymoar :< .

eduardo579, 2017-02-24:
With beta 4, I have this issue again. In beta 3 it was less frequently. But now, in 6 hours I had to reboot twice due to this.

gloddo, 2017-02-25:
Yeah I can confirm it still happens at least once per day (the video thing). About sound it didnt happen anymore since I uninstalled VIPER and Dolby, but now speaker volume is much lower than what it used to be in flyme

Moyster, 2017-04-01:
Should be fixed in next build, feel free to re-open if Beta 8 still have this  bug ;)

closed 2017-04-01 by Moyster