Moyster/nougat_device_meizu_m2note #16: Battery drain / no deep sleep

Ricky392, 2017-02-26:
After flashing the rom and use it for a while, the telephone can't go in deep sleep because of the "EINT Wakelock".
I tried everything, from a clean install, switching off wifi, turning off google hotword detection and so on, but nothing worked.

I'm attaching some logs I hope you found useful, if I can do something specific to help you, let me know !

https://drive.google.com/open?id=0B2fa1zm_NO5xM1pjeHBkcDFNM00

Ricky

Moyster, 2017-02-28:
Hi thanks for your logs, very detailed, I'll give this a look later this weekend :)
In the mean time can you try this kernel and report if that's any better regarding deep sleep ?  
https://drive.google.com/open?id=0B6BtEmoqqQhcUWJZNnVZNTBiZzg  
From twrp : install > flash image > select "DPboot.img" > boot > flash > reboot > ??? > profit

Ricky392, 2017-02-28:
Tried to flash your kernel but nothing change... I provide you screenshots + new log
Thanks for your help, I'm putting effort in it because I really belive in your work !!! 
Countinue like this!

https://drive.google.com/open?id=0B2fa1zm_NO5xeXJzemtIa25sUjA

Ricky392, 2017-03-01:
[image]

An update , it goes in deep sleep but I don't know why. In this moment when I'm writing to you it goes back to "no deep sleep"

Moyster, 2017-03-01:
Allright, new kernel : 
https://drive.google.com/open?id=0B6BtEmoqqQhcS3pYTHR4cjY0WVk

Can you also post the output of :  
(adb root && adb shell) : "cat /proc/interrupts |grep eint"  
  
Let the phone go idle and repeat the cmd a few times over 1-2min to see if when Screen-Off there is any value that grows.  
Only lsm6ds3 does a lot of interrupts but never when Screen-Off (which is normal)

Ricky392, 2017-03-01:
Hi I flashed the latest kernel you gave me but no luck.
I check the log file you asked me and this is the result:
[image]

I made a quick research and I found that is related with sensors... but I think you already know it ! ;-)

Moyster, 2017-03-01:
Yup, seems like lsm6ds3 is tripping balls... I modified the driver a bit on the second kernel I linked.  
Does the number increase if the screen is off ?
Did you enable features like "ambiant display" or the "wave hand" gesture ?  
I've seen Meizu updated their m2 note kernel in flyme 6, it was 3.10.7x instead of what they released initially (3.10.65), it might be worth to ask them for updates here :  
https://github.com/meizuosc/m571  
I guess an issue asking nicely would be enough ;)

In the meantime, I'll keep looking for a solution, btw is your device a m571 or a m571H ?

Ricky392, 2017-03-01:
Yes, it was activated, but when i realized the sensor related issue, I deactivated the "ambient display" feature. now I have to test with it deactivated.
To get little deeper, I made a simple service that tracks accelerometer data during time and I found that with AD activated, the service recive data even during sleep, but when I disable it (and reboot the phone), during screen off I do not recive data. So I think this is the major problem. I will confirm you in the next days with few more testing.
Thanks !
PS m571H

Ricky392, 2017-03-02:
I finally discoverd the culprit... it's **WHATSAPP** . If I simply remove it from the recent apps, mt-eint lsm6ds3 stops growing.
I already wrote to WhatsApp team, let's see next

Moyster, 2017-03-02:
Good ;) That's supposed to be android (frameworks) behavior, to turn most of the sensors off when screen-off ( we already use meizu's kernel, but they didn't release their recent updates, we have Flyme 5.1.8.0 ~~ sources, Flyme 6 kernel sources are not released yet, maybe they improved a few things regarding our hardware).  
It's kind of wierd Whatsapp needs a gyroscope :D but that app is known to be a pain in the ass for a lot of android rom/users.

Moyster, 2017-04-01:
Well, I'll consider this closed, ambiant displays is known to cause battery drain :D

closed 2017-04-01 by Moyster

Ricky392, 2017-04-01:
ok, yes the real battery drainer is AD.
Just want to understand why the kernel does not stop the sensors when locked

Moyster, 2017-04-01:
It does, AD keeps them on for things like "pickup to wake", it uses the accelerometer to detect you pick up your device in hand, and it doesn't play well with our device it seems :D

Ricky392, 2017-04-01:
yeah yeah I do understand, but I don't about what's happening with whatsapp ahah
fucking whatsapp

Moyster, 2017-04-01:
does whatsapp vibers on notification ? It might be enough to trigger the sensors, idk ^^, whatsapp seems to be an endless source for troubles on 3rd party roms :)

Ricky392, 2017-04-01:
Actually is something related with the main activity (i made a reverse engineering of the app) it's happening since the last update with new features of camera and status