Moyster/nougat_device_meizu_m2note #18: if (dual sim && enable gps && use some navigation app): lose sim card signal

tt677666, 2017-03-02:
if  (dual sim card && enable gps && use some navigation app): 
       {
           sim card offline
       }

#thank for your working

Moyster, 2017-04-01:
It might have changed, try with latest build (beta 7 "IsNoJoke")

Moyster, 2017-04-17:
Well, if this is still happening, feel free to reopen the issue :)

tt677666, 2017-05-21:
@Moyster  

i try it agin with version "lineage-14.1-20170430-UNOFFICIAL-m2note.zip",found  the bug still unsolved 
and i collected log with catlog hope to help you.
thx!

Moyster, 2017-06-22:
I'll wait for wileyfox spark/spark+ update to nougat to update our GPS drivers (with Sparks blobs), it's hard for me to get 2 simcards and test.
If you need this and want to help, I guess you could try using blobs from S3plus (from MAD team, they updated gps to N already) but I'm pretty sure it won't work without android-source sided changes.

ps: wileyfox spark(+) should be updated before end of june, so... fingers crossed.

Moyster, 2017-07-11:
It was to be expected that Wileyfox are crooks and liars, Their mediatek devices will only get updated Late July (read. they will release an update around October 2017, the very same bugged June 2017 update, just... 6 months after so they're sure their Sparks users have moved on and nobody will notice.  
Will see if N blobs from already released roms are properly working.

closed 2017-11-11 by Moyster