Moyster/nougat_device_meizu_m2note #21: Chrome Crash

undaclazz, 2017-03-17:
Hello Moyster

I don.t know if it's Chrome faulty or ROM faulty, I tried to submit info to Google but they don't reply.
Chrome crash everytime I try to share a web page to Facebook with click a button, and it happened since Build 5 .

`03-16 07:03:37.961 E/BufferQueueProducer(382): [SurfaceView - com.chrome.dev/org.chromium.chrome.browser.ChromeTabbedActivity] dequeueBuffer: BufferQueue has no connected producer
03-16 07:03:38.037 E/BufferQueueProducer(382): [SurfaceView - com.chrome.dev/org.chromium.chrome.browser.ChromeTabbedActivity] dequeueBuffer: BufferQueue has no connected producer
03-16 07:03:38.063 E/chromium(19664): [ERROR:gles2_cmd_decoder.cc(15365)] Context lost because SwapBuffers failed.
03-16 07:03:38.063 E/chromium(19664): [ERROR:gles2_cmd_decoder.cc(5279)] Error: 5 for Command kSwapBuffers
03-16 07:03:38.120 E/chromium(19664): [ERROR:gles2_cmd_decoder.cc(4238)]   GLES2DecoderImpl: Trying to make lost context current.
03-16 07:03:38.124 E/chromium(19664): [ERROR:gles2_cmd_decoder.cc(4238)]   GLES2DecoderImpl: Trying to make lost context current.
03-16 07:03:38.183 E/chromium(19664): [ERROR:gles2_cmd_decoder.cc(2475)] [.Offscreen-MainThread-0xce746000.CmdBufferImageTransportFactory-0xcc851200]GL ERROR :GL_OUT_OF_MEMORY : BackFramebuffer::Destroy: <- error from previous GL command

Thanks and sorry If I posted on wrong section.`

Moyster, 2017-04-01:
Sorry for the long delay, I think this might be related to how/what webview LineageOS is shipping.
Currently lineage ships with "aosp webview" and they are working on a "chromium-webview" fork.
It might be related to your problem and you should try installing "google webview" :  
https://play.google.com/store/apps/details?id=com.google.android.webview&hl=fr

Moyster, 2017-04-01:
ps: I closed this as Lineage will move on to switch on their chromium fork and since then, installing google's one should fix it ;)
ps2: you might want to check in Settings > Dev options > Webviews and select the newly installed webview (or chrome's if it appears)

closed 2017-04-01 by Moyster

undaclazz, 2017-04-01:
ok.. thanks Moyster . :)