Moyster/nougat_device_meizu_m2note #30: I hope for Your help.

teik-off, 2017-04-19:
LineageOS-14.1 Build 11 (RC 1 "Anougat") revealed the following drawbacks:
- does not work bluetooch;
- does not work manage root accesses;
- vibration does not work during the answer and at the end of the conversation.
[logcat.txt](https://github.com/Moyster/nougat_device_meizu_m2note/files/932290/logcat.txt)
[image]
[image]
[image]

Moyster, 2017-04-19:
Hi, thanks for that.
Well I triple checked BT, it should work (I use a smart watch daily, every features work so far), check if your bluetooth have a MAC adress in the settings. try cleaning cache apps of "bluetooth services", then a clean install.  
Manage root access : since I enabled enforcing, lineage build makes it "grey", if you flash superSu, you will have root, if you use "adb shell" then type "su", you will have adb with root. 
The vibration on call/hang up, I don't know, was it working before ? I'll see what I can do for that.

teik-off, 2017-04-19:
I mean regular root, which can be obtained by using addonsu - that it doesn't work (root access does not grant the application).
If you put the supersu - root work. But why reinvent the wheel, if the developer has you covered.

teik-off, 2017-04-19:
Bluetooth MAC address is not available. The firmware was installed several times. Always with full wipe.

teik-off, 2017-04-19:
The vibration in the contacts stopped working with beta 10.

teik-off, 2017-04-19:
Thank you for your work. Maybe you need this link
https://4pda.ru/forum/index.php?s=&showtopic=761874&view=findpost&p=60595918

Moyster, 2017-05-02:
Hello, Is vibration working (the pickup/hangup call, etc?) on the latest build ?
If it stopped with build 10, it might not be related to selinux, but I might be wrong.

Regarding root, you know lineage doesn't build roms with root inside anymore ? It's considered a bit of a security issue, and flashing supersu allows to have root for those who wants it (and those who don't want root / don't know what it is, dont have it enabled by default).
There's an option to restore root options in lineage, but there is actually people who don't use root ^^ (me the first).

teik-off, 2017-05-03:
Hey. You have done a great job. LineageOS-14.1 Build 12 (RC 2) has visibly improved. That nemesis is the following bugs:
The vibration does not work at the answer (but when the call ends all works), then half the problem you decided;
The normal superuser mode does not work (it is not active).
In the rest everything is super! Continue in the same direction.

P.S. Yes, I know that not everyone needs superuser mode. But why can not you make it active? In order to be able to install addonsu and use the regular superuser mode https://lineageos.cc.kremowka.xyz/extras
I'm not an supersu supporter. Since the developer provided this and made the addon as addonsu.
Yes, I know that the super-user mode is not provided in LineageOShttp: //www.androidpolice.com/2017/01/20/lineage-os-official-builds-coming-weekend-data-migration-builds-available-cm-users/

teik-off, 2017-05-03:
I for that all was regular. If there is an option provided by the developer to get the regular superuser, why not implement it. Every man must decide whether to put him addonsu or not.

Moyster, 2017-05-11:
I don't really think I understand, but as of now, building lineage as user/signed-keys doesn't allow adb as root, nor enabling root for apps, and it doesn't pack "su" binary.
"addonsu" does the same thing as "supersu", supersu is closed source, "addonsu" is lineage's open-source "alternative", so it should do the same thing, the UI element is greyed because the build is user/signed.
Anyway, I didn't "do it on purpose", but the way it is actually is fine, "addonsu" and "supersu" are kind of outdated, and systemless root is a far better approach (magisk and phh supersu).

If you really want to ungrey and use these options (instead of supersu / magisk), you can download my kernel sources / device tree / lineage 14.1 sources, all you need to do is build boot.img as userdebug :  
. build/envsetup.sh && breakfast m2note
make bootimage -j4

It shouldn't take much time (between 2min and 15-20min depending on your pc) and the userdebug boot.img will be in android_sources/out/target/product/m2note/(boot.img). 

(on another note, I still haven't closed this issue because vibration is still not solved, I edited the topic name for clarity)

teik-off, 2017-05-25:
Hey. You did an excellent job. LineageOS-14.1 Build 13 (RC 3) has improved noticeably and has become very fast.
Vibration has worked, you can close the topic.
Thank you for your work.

P.s. Remove the entire "Manage root acceser" item from the Developer options settings, it is still not active (it only flashes in the eyes).

Moyster, 2017-05-25:
Haha, well, regarding vibration, thanks goes to lineage ;) (seems the call duration bug is fixed also)

Ps: but... I use that menu when I need to flash debug kernels :D

It would be nice actually if someone could for example, build a userdebug/enforcing kernel (so root in dev options are allowed) and a userdebug/permissive one if people really wants it :)

closed 2017-05-25 by Moyster