Moyster/nougat_device_meizu_m2note #31: GPS smart lock

micheleberrettis1, 2017-04-21:
Location Smart-lock is not working for me... Tried to wipe system and reflash but it doesnt work anyway...

Moyster, 2017-05-11:
Hello,  
What doesn't work in smart-lock, verified locations ? verified devices ? voice ? 
What do you use, a pin lockscreen, a password/scheme unlock, etc etc...
Did you make sure you allowed "smart-lock" as a trusted app or something...

I have no clue if it can be fixed easily or not, but it took 20 days for me to care about this issue, or should I say single line with absolutely no details, no logs. and I only did because I answered/closed other issues.
I'm still wondering why I do all this, the more it goes, the less content there is in issues. If you know about posting github issues, you probably read my "little guide" on "how to post a github issue", so you know I want details and/or logs if possible, now here's me being (all of) you :  
"My car has a problem, it does a wierd noise, sometimes it doesn't, but when it does, I don't know if I was rolling for a long period or not, can you fix it ?"

micheleberrettis1, 2017-05-11:
Reflashing rom fixed the issue...

Moyster, 2017-05-11:
Yes, I was just testing this, I set home as a trusted place, set a pinlock, locked my phone, unlocked it : "your phone is unlocked !" with a small explanation about smart-lock, now since meizu is shit and they've done voodoo to the sensors our devices have, maybe the "detect if the device is carried and stay unlocked" might not work. Blame meizu.  
Apart from that, it seems it works, case closed :)

closed 2017-05-11 by Moyster