Moyster/nougat_device_meizu_m2note #36: I found a bug

SUtong95, 2017-05-11:
open "disable all cellular data access"
The phone will always restart！

Moyster, 2017-05-11:
Hi there, your bug issue is really incomplete, I don't own a crystal ball and I'm not even sure what you're talking about.
I tried this : Settings > Data consumption > Apps data usage > select an app (I tried with youtube / android system / google play).
There's options here to disable specific apps to access mobile data, "disable all cellular data access", if I enable this, nothing happens, the apps can't browse internet on mobile data but it does on wifi, and I have no reboot.  
If you have steps to reproduce the bug, then make logs out of it, save and upload them so I can check it.  

Did you follow the quick advice on my Xda thread, ie. doing a clean install and so on ? Are you using mods ?  99% of the time these issues are just me "assisting people in properly instaling their rom".

Feel free to reopen if this still happens after checking the problem is not on your end.
ps: this should be the first thing to do when you find a bug nobody else have.

Moyster, 2017-05-11:
Well, I did try but it didn't crash at first. I couldn't reboot my phone with wifi disabled tho. my bad :D  
As a small guide to troubleshoot this :  
If your phone bootloops because you enabled "disable all cellular data access" on any apps, remove your simcard and boot the phone, turn on wifi, put your simcard in, then disable the "disable all cellulaar data access" options on every apps you enabled it on.

m3shat, 2017-05-25:
I experience the same bug; This exception gets thrown, maybe it helps

`05-25 22:44:05.112 15406 15406 D AndroidRuntime: Shutting down VM
05-25 22:44:05.113 15406 15406 E AndroidRuntime: *** FATAL EXCEPTION IN SYSTEM PROCESS: main
05-25 22:44:05.113 15406 15406 E AndroidRuntime: java.lang.RuntimeException: Error receiving broadcast Intent { act=android.net.conn.CONNECTIVITY_CHANGE flg=0x4000010 (has extras) } in com.android.server.NetworkManagementService$2@4e0f74a
05-25 22:44:05.113 15406 15406 E AndroidRuntime: 	at android.app.LoadedApk$ReceiverDispatcher$Args.run(LoadedApk.java:1132)
05-25 22:44:05.113 15406 15406 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:751)
05-25 22:44:05.113 15406 15406 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:95)
05-25 22:44:05.113 15406 15406 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:154)
05-25 22:44:05.113 15406 15406 E AndroidRuntime: 	at com.android.server.SystemServer.run(SystemServer.java:383)
05-25 22:44:05.113 15406 15406 E AndroidRuntime: 	at com.android.server.SystemServer.main(SystemServer.java:237)
05-25 22:44:05.113 15406 15406 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
05-25 22:44:05.113 15406 15406 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:889)
05-25 22:44:05.113 15406 15406 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:779)
05-25 22:44:05.113 15406 15406 E AndroidRuntime: Caused by: java.lang.IllegalArgumentException: command '79 bandwidth addrestrictappsondata ccmni0 10113' failed with '500 79 Unknown bandwidth cmd'
05-25 22:44:05.113 15406 15406 E AndroidRuntime: 	at com.android.server.NetworkManagementService.restrictAppOnData(NetworkManagementService.java:2005)
05-25 22:44:05.113 15406 15406 E AndroidRuntime: 	at com.android.server.NetworkManagementService.processPendingDataRestrictRequests(NetworkManagementService.java:1971)
05-25 22:44:05.113 15406 15406 E AndroidRuntime: 	at com.android.server.NetworkManagementService.-wrap13(NetworkManagementService.java)
05-25 22:44:05.113 15406 15406 E AndroidRuntime: 	at com.android.server.NetworkManagementService$2.onReceive(NetworkManagementService.java:382)
05-25 22:44:05.113 15406 15406 E AndroidRuntime: 	at android.app.LoadedApk$ReceiverDispatcher$Args.run(LoadedApk.java:1122)
05-25 22:44:05.113 15406 15406 E AndroidRuntime: 	... 8 more
05-25 22:44:05.113 15406 15406 E AndroidRuntime: Caused by: com.android.server.NativeDaemonConnector$NativeDaemonArgumentException: command '79 bandwidth addrestrictappsondata ccmni0 10113' failed with '500 79 Unknown bandwidth cmd'
05-25 22:44:05.113 15406 15406 E AndroidRuntime: 	at com.android.server.NativeDaemonConnector.executeForList(NativeDaemonConnector.java:502)
05-25 22:44:05.113 15406 15406 E AndroidRuntime: 	at com.android.server.NativeDaemonConnector.execute(NativeDaemonConnector.java:401)
05-25 22:44:05.113 15406 15406 E AndroidRuntime: 	at com.android.server.NativeDaemonConnector.execute(NativeDaemonConnector.java:396)
05-25 22:44:05.113 15406 15406 E AndroidRuntime: 	at com.android.server.NetworkManagementService.restrictAppOnData(NetworkManagementService.java:2002)
05-25 22:44:05.113 15406 15406 E AndroidRuntime: 	... 12 more`

Moyster, 2017-06-05:
fixed in the upcoming release :) (rc4, should be online in less than 30min)

closed 2017-06-05 by Moyster