Moyster/nougat_device_meizu_m2note #39: Downloading files in the browser does not work

teik-off, 2017-05-29:
При попытке загрузить файл на сайт (любой сайт), пишет что файл не выбран.

teik-off, 2017-05-29:
Installed LineageOS 14.1 build 13, plus RC3-memfix-boot-v2.img.

Moyster, 2017-05-29:
uploading or downloading ?
Like, going to imgur to upload a photo ?

teik-off, 2017-05-30:
I can not upload dailies (photos, videos, apps) on a site.

teik-off, 2017-05-30:
Yes, I can not download (I conducted an analysis, this problem is not for me alone).

Moyster, 2017-05-30:
Jello can't upload for some reasons, I tried firefox and it worked fine, blame Lineage :P

teik-off, 2017-05-30:
I think they will fix this problem.

Moyster, 2017-05-30:
Yep, seems like Jelly is WIP :  
https://review.lineageos.org/#/q/project:LineageOS/android_packages_apps_Jelly

Use ff or chrome during that time ;)

teik-off, 2017-06-04:
Hi, assemble for a change some other firmware. Demand will be, and users need a variety.

Moyster, 2017-06-04:
I don't have time for that, anyone is free to use what's on my github to start their own project.  
There's a million things to fix and improve in Lineage and I'm not familiar with other roms I'll have to troubleshoot to get it to boot again (because Lineage includes more "old devices" compatibility stuff than others, let say, CarbonRom or whatever.)  
And I have no "magic patches" to release, if anyone wants custom roms, they can use their brain, read compilers errors, fixes them.

teik-off, 2017-06-04:
Here's a nice firmware http://aicp-rom.com.
Using your git is easy. Except for the camera, it does not work.
You can try to collect for the sake of experimentation.

teik-off, 2017-06-04:
https://github.com/AICP

teik-off, 2017-06-04:
I understand that you do not have time. Therefore, no one insists, all according to the will and desire.
Thanks anyway for your work.

teik-off, 2017-06-06:
Hey. Judge yourself. The life cycle of the phone is about 2-3 years. In the official sale of this phone is no longer. Except for the online store hthttps: //www.aliexpress.com. Soon all those who used it will switch to other phones, and as I think it will not be meizu's phones. I think users deserve a variety of firmware. If I could understand the reason for the inoperative camera, I would have fixed it for a long time already, because the main reason for which it is impossible to use other firmware is this.
Yes, most users are not grateful and do not appreciate the work of others. Nothing can be done about this.
But I think the other half of users will appreciate the variety of firmware and there will be demand for them.

Moyster, 2017-06-06:
Yes, I'm judging myself. On cm-12.1 I was working on my github, and I was making a lot of pull requests to some "CM-M571" group, I had no idea who this was.  
Then pull requests stopped being accepted, and it seemed inactive so I continued alone.  
After that some(one/group) started building CM-12.1 from where "cm-m571" was left, "this rom is better than moyster's", "this rom is 100% no moyster", "separate projects".... Cool.  
Now, AlexLartsev started doing his "blood kernel", telling me I was bad, I sucked, his kernel was better...  
I never meant to "fight against others" so I tried to 'calm' the situation, asked why he was telling me all that, "I'm bored" he said... he's insulting people because he's bored...  

Since I didn't want to work ALONE on marshmallow or Nougat, I offered him to help and ask anyone he knew to JOIN US. This was October 2016. NOTHING HAPPENED. Nobody joined, Alex did nothing, he had access to my repositories and I told him even the smallest help counts (you know about Repka ? I was told it's a great story about contributing and working together).  

Now fast forward 4 months later, getting insulted everyday because "alex" is bored, (he wrote "blyat" and "are you ahuel tam" like, 357 times during December, while asking me to do him "test builds" of nougat everytime I did progress (XD).  
Where is help ? It's funny to see I'm being insulted and called a selfish person when nobody  realise there would be NO ROM for this phone, if I didn't try hard and got some help... Who knew it was even possible ? Not me. Exactly, nobody believed in or trusted me, so nobody cared to do ANYTHING. "wait moyster, he did cm-12.1, he will do it again...".  

This summarize this community and why I don't share anything, lazy uneducated people just waiting to  get their hands on "ready to build" projects, I was the first, not because I'm the best, but because I'm the only one who tried, and my decision would be different if anyone else had TRIED SOMETHING, instead of just waiting for me, begging, harrassing, insulting, then apologizing and begging again.

I was spammed and told so many times I'm worthless by exactly the same persons who wants to build "alternative" roms for m2note, that it's just IMPOSSIBLE to help anyone now, they just did too much, and releasing anything will make them win.  
Here's a thing, he, his friends and I don't even know who (since people come to insult me with "new accounts" on Xda xD) can suck my dick, if they thought being stupid, uneducated and disrespectful lead to anything of value. There's nothing for them, from me, simple.  
Look what happens when I just release something and I don't have time to upload my device tree : https://forum.xda-developers.com/showpost.php?p=70257575&postcount=18  
And then, that very same person (dmitrywxz) started spamming me and asking for IMMEDIAT RESULTS and me "fixing my mistakes" or not uploading my patches, patches I found on google (??) and that I was nothing, nobody, etc...

Now, I don't really care that the phone will "die" soon and people switch, I mostly did it for my own sake, I learned my lesson, made some people happy, and I won't "comply" to people who's best argument is "you're selfish" when their own behavior lead them to having nothing.

If you read that far, you have to know, there's a lot of approaches to fix the camera, one of them is done by "shimming" some camera functions that is crashing because it changed on MM/Ng, and it's on lineage gerrit. But I guess it's easier to insult me than trying and pulling some work. Because up until now, my device tree contains most of the "patches" converted to shim, so there _IS_ no patches to apply.  
This was my initial goal, providing a device tree without patches because slowly reimplementing other's shims and/or making my own. I guess I'm legitimately selfish right ?? (and I already told Alex all of this, which he seems to ignore).

tl;dr: if you want free vegetables, don't insult the farmer, or start growing plants.

Moyster, 2017-06-06:
And i have to admit, I'm still amazed those persons after being so cruel and mean, thinks I will _ever ever ever ever_ release anything that helps them.  
Internet is not a place you should act different than in real life, y'all won't go harassing a baker for free breads, you won't stay in front of the bakery to insults customers, etc...  
Why don't they start working on it ? Gathering help or resources/infos, on how to solve that camera issue ? Why don't they harass MAD TEAM instead ? Or Danielhk ? Or divis ? **They all are way better devs than I am.**  I guess it's easier to insult people you thinks are your peers. wrong, I'm far away from being like them.

closed 2017-06-06 by Moyster

teik-off, 2017-06-06:
I did not quarrel with you. Bad that it happened, I repeatedly burned at this. I will say so, AlexLartsev did not help anyone, he tried to make a kernel. The rest of the help from him nebylo. The fact that you have a conflict I expected, judging by the forum on xda.
Dmitrywxz is the only one who knows this, but he was powerless and could not start the camera.
And I do not have enough experience to collect firmware, time and capabilities.
So, okay, I'll be content with what's there.
Thank you all for the work !, and firmware.