Moyster/nougat_device_meizu_m2note #43: Camera close version 15

javijdd, 2017-06-18:
[image]
[image]

kosmosofby, 2017-06-22:
Same problem

Moyster, 2017-06-22:
Hi there, There's a solution on the Xda thread, some people deleted a file and got camera working back, I'm not exactly sure it works (but a reliable tester tried and reported it works fine).
Deleting the "CAMERA_VERSION" file should fix it, search for that on Xda and read the comments/feedbacks and make a backup before doing anything ;)

closed 2017-06-22 by Moyster

kostasil, 2017-06-24:
Hello guys. Where can i find this file?

javijdd, 2017-06-24:
CAMERA_VERSION delete i camera working 100% thanks