Moyster/nougat_device_meizu_m2note #45: Idea for new feature

janek125, 2017-06-23:
Maybe od possibile to make something to turn off and turn on phisical homebutton?
My homebutton is broken, sometimes itself take back mamy times. This close all of my applications.

Moyster, 2017-07-11:
Hi, It was previously requested back on CM-12.1, the thing is to make the touch button inactive, someone has to modify our kernel and add a sysfs node to enable/disable the key, which then can be used by Android to enable/disable it.

Any help is welcomed, I massively use the touch button so I'm not trying.

janek125, 2017-07-18:
Hi 
Thanks for interest. I solved the problem to half. I edited /system/user/keylayout/mtk-tpd.kl and insert # before line: key 102 BACK VIRTUAL. It solve my problem with many times "click" back. Now my home button isn't work when I touch it. Still work when I push it but it's OK for me. 

A few data ago I'm install Resurrection Remix OS 5.8.3 and there is possible to turn off phisical button completly. Now I'm back to your LineageOS because Resurrection Remix OS 5.8.3 drain my battery a lot.

micheleberrettis1, 2017-07-31:
There Is a magisk module to turn off phisical button on LineageOS

Moyster, 2017-08-01:
Not a long term solution tho, it needs magisk and a module, and as it only prevents the actions sent by the button to be parsed and trigger anything in android, (broken) physical button will still spam key events and act retarded/broken.  
The best/only proper option is to add an enable/disable node to the kernel driver (which might never happen due to lack of resources)

maTyaMusic, 2017-10-31:
Add camera2 API :)

Moyster, 2017-11-06:
you can try camera2 api for yourself, if our device supports it and meizu/mtk built/compiled the blobs with it, add "persist.camera.HAL3.enabled=1" to build.prop, reboot and test camera2, if it works, it's cool, if it doesn't, I can't change that.

closed 2017-11-06 by Moyster