Moyster/nougat_device_meizu_m2note #47: Pokémon Go

ErenTitan, 2017-07-01:
When I open the game give me an error: GPS signal could not be detected.

ErenTitan, 2017-07-02:
I have high precision activated.

Moyster, 2017-07-02:
Works fine, make a backup and do a clean wipe ;)

closed 2017-07-02 by Moyster

ErenTitan, 2017-07-02:
I did but it still gives me the same error but Google Maps works. What can I do?

Moyster, 2017-07-02:
Try "gps test" app, if it gives you "GNSS status : 3d fix" it means gps works properly.  
Make sure you don't have root enabled / flashed mods / etc.. and reinstall Pokemon Go clean from the appstore.

ErenTitan, 2017-07-02:
@Moyster I install android 5.1 and now works but in 7.1.2 didn't detect Gps Signal. Whan can I do?

ErenTitan, 2017-07-02:
The gps works fine because I flash the Cyanogenmod 5.1 Rom and the game didn't give me any error. Now I put android 7.1 and I'll try to do what you told me.

ErenTitan, 2017-07-02:
Yes!!! Now It works well thank you @Moyster!