Moyster/nougat_device_meizu_m2note #49: SeLinux policy

eduardo579, 2017-07-31:
I was wondering if you can set SeLinux to permissive by default for next builds. If not, is there any method to change it? I want to flash DivineBeats due to the sound problem.

Thanks and enjoy your summer!

micheleberrettis1, 2017-07-31:
To change SElinux: root, download from xda (not playstore) HEBF, look for the option to change SElinux...

micheleberrettis1, 2017-07-31:
It should Be on the tools section

eduardo579, 2017-07-31:
I've installed the app but it is not showing the option. Maybe it's because I'm on RR and it has an option to change SeLinux in Configuration. Nevermind because the toggle doesn't change it.

Moyster, 2017-08-01:
https://forum.xda-developers.com/android/software-hacking/mod-kernel-permissive-patcher-android-t3506338  
This kind of tools might work, but there's no real reasons to turn off selinux globally, selinux prevents a lot of "bad behaviors / risky stuff" to be used/accessed from.  
There's an "easy way" to add support for mods to enforcing selinux, compile boot.img yourself with selinux=permissive flag, use logcat to grec selinux denials, then use audit2allow / your brain and SL tools to fix the denials, rebuild boot.img without permissive flag, divinebeats works in enforcing.  
Now, I don't really support mods or anything outside of lineage, so if you need it, you can do it 👍  ;)

closed 2017-08-01 by Moyster