Moyster/nougat_device_meizu_m2note #51: After Everything Done, Camera is not working and the Flash light is not responding.

droidbuzzer, 2017-08-04:
Hi,
i have recently rooted & unlocked my phone for the first time, followed instructions correctly. Phone is working well, even the call quality is increased but the huge disappointment is the 
1. the flash light in the status bar is not responding.
2. Camera App is not opening at all. it says its not responding.
i have not installed any other mods with the lineage OS 14.1(Latest)

dariusworks, 2017-08-05:
#48

bumperboy, 2017-08-06:
Hi everybody. I have the same issue and even after applying the camerafix.zip patch suggested by dariusworks, the issue is still there.

Moyster, 2017-08-06:
https://github.com/Moyster/nougat_device_meizu_m2note/issues/43

Read the previously closed issues about camera, you don't need the patch.zip (it only changes cameraserver permissions to "system" which is harming the device's security).

Deleting the file CAMERA_VERSION in your nvdata partition and rebooting will do. Someone can make a flashable zip to delete that file (I asked if anyone could do it but I don't think anyone did anything except applying to its own device).

update.zip > add "rm /path/CAMERA_VERSION" > mention it here > the patch will be pinned on xda  

(ps: thanks @dariusworks for trying to help ;) )

closed 2017-08-06 by Moyster

Dinolek, 2017-08-14:
@Moyster Flashable fix for camera
https://github.com/Dinolek/asa/raw/master/CameraFixLineageos14.1.zip

Moyster, 2017-08-14:
finally, THANK YOU :) will update the thread over xda

Dinolek, 2017-08-14:
@Moyster How i can contact with you?

Dinolek, 2017-10-24:
[NewCameraFix-Dinolek-Los14.1.zip](https://github.com/Moyster/nougat_device_meizu_m2note/files/1412365/NewCameraFix-Dinolek-Los14.1.zip)
I changed old path for mounting to /dev/block/mmcblk0p21, so now will work on recovery with new and old path.