Moyster/nougat_device_meizu_m2note #55: Facebook crash only data network

javijdd, 2017-08-30:
Hi, I have this problem only with the data of the network the facebook is stopped a message comes out the solution that can show the thanks

Moyster, 2017-08-31:
How do you know it's facebook and what crash are you talking about ? You don't submit logs, I don't have no crystal ball.

javijdd, 2017-09-01:
ok you should on when you are browsing facebook with the network data i stopped at the moment you have to close it but before closing it you see that it is stopped and a message comes out and it says close the app by the app i have to close it to work but it does not happen with the original rom if it is not rom and it's just facebook when it happens I send you a photo does not always happen

javijdd, 2017-09-02:
[image]