Moyster/nougat_device_meizu_m2note #56: Unknow contact in call history

Ricky392, 2017-09-06:
Dear Moyster, I found a bug in the call history.
When I make a call, if it last more than some seconds, the call log shows "unknown" instead of the contact detail.
I tried to wipe the rom, force close the telephone app, clear data and cache, swap the sim on the 2 slots, check if it is something related with the saving position of the contact, but nothing of that works.

how can I make a log of this issue ? do you think is there a particular moment to do the log?

thanks

Riccardo

Moyster, 2017-09-06:
Hmm, it's wierd I don't have this issue here (calling another number / my voicebox, for 10sec as well as 30+min) is your number supposed to be displayed but it shows "unknown" ? If that's the case, you can go in your sim settings and define your own number (it's needed for multigroup messages as well).  

If that's not, are you sure your network operator / yourself didn't disable "number presentation" ? That seems more likely than a rom bug (I don't really read the thread on xda, so I don't think it was reported by anyone else).  
There was a previous Italian user (if I remember you are too) who had troubles with calling a service number to reload his prepaid phone and he ended up calling the "carabinieri" (?spelling).  
So, let's try and troubleshoot this, I'm not sure if logs could be relevant here, and I can't teleport to italia.  
Did it work previously ? Does it work in another phone / flyme stock rom ? Check your operator account options in case for some reasons the "number presentation" option was disabled.  

Last but not least, I'll build another lineage-14.1 rom in a few days, I'm just waiting for the usual security patch bulletin to be merged, I didn't really touch anything phone related but maybe that was a lineage/aosp bug & it got fixed by now. We'll see ;)

Ricky392, 2017-09-07:
No it's not related with my ita carrier, nor with the "number presentation" thing. It happens just with this rom. 
I found that if it's the first time a contact enters in the call log, everything is ok, but if i call him/her again, an unknow entry appear in the log after the contact details.

ahah you spelled carabinieri well ! xD

off topic, what about the low volume ? I tried to edit some files in the system but nothing helps -.-

Ricky392, 2017-09-22:
Hi Moyster, I tried to change the sim number and etc but nothing works... I tried to connect the phone with ADB and found strange thing when closing the phone call.
Hope this could be helpful ! thanks
[log_unknow_calllog_issue.txt](https://github.com/Moyster/nougat_device_meizu_m2note/files/1324146/log_unknow_calllog_issue.txt)

Moyster, 2017-09-24:
try the latest version of the rom and tell me if that's still a problem or not ;)

Ricky392, 2017-09-24:
Thank you very muchhh ! trying immediatly !!!
I'll let you know asap ! :-)

Ricky392, 2017-09-24:
Unfortunately it still says "unknown" and even the logs are the same as above.
Really don't know what to think... this bug is very annoying...

closed 2017-09-24 by Ricky392

Moyster, 2017-09-24:
xD no problem, I knew it was probably due to some network wierdness, if you have a valid fix, post it here it might help people :)