Moyster/nougat_device_meizu_m2note #62: NVRAM Warning

pedgsilva, 2017-12-05:
My WIFI display NVRAM WARNING:Err=0x10

I've tried to find the directory data/nvram to substitute the wifi file but i can´t enconter the directory to do this : https://forum.xda-developers.com/showthread.php?t=2730486

Thank you

sandstranger, 2018-01-03:
this is just warning
this commit from MAD team will hide this warning
https://github.com/fire855/android_external_wpa_supplicant_8-mtk/commit/76506048d037351d14a64145aadc0c43ae6ce7ac

Moyster, 2018-03-04:
Yeah, I'm still wondering if it's "safe" to hide such warnings, because clean nvram files shouldn't show a "nvram warning:err=0x10" entry, https://forum.xda-developers.com/showpost.php?p=74406732&postcount=159 this post seems to indicate if /data/nvram doesn't exist, you should try /nvdata ( our devices uses /nvdata/ as far as I know, but it's linked to /data/nvram, so both should work, make sure you're root / using a root explorer)

closed 2018-03-04 by Moyster