Moyster/nougat_device_meizu_m2note #64: Build Error

micheleberrettis1, 2017-12-27:
`frameworks/av/services/audioflinger/Threads.cpp:6466:29: error: no member named 'get_capture_position' in 'audio_stream_in'
        if (mInput->stream->get_capture_position != nullptr
            ~~~~~~~~~~~~~~  ^
frameworks/av/services/audioflinger/Threads.cpp:6469:39: error: no member named 'get_capture_position' in 'audio_stream_in'
            int ret = mInput->stream->get_capture_position(mInput->stream, &position, &time);
                      ~~~~~~~~~~~~~~  ^
2 errors generated.with
[ 49% 27633/55571] target arm C++: libaudioflinger_32 <= frameworks/av/services/audioflinger/AudioMixer.cpp
ninja: build stopped: subcommand failed.
build/core/ninja.mk:151: recipe for target 'ninja_wrapper' failed
make: *** [ninja_wrapper] Error 1

#### make failed to build some targets (04:13:41 (hh:mm:ss)) ####
`
Getting this error when building stock lineage without any modification.

Dinolek, 2017-12-28:
This issue was already there, just check next time.
You need to remove get_capture_position function from frameworks/av/services/audioflinger/Threads.cpp

Patch for it:
https://github.com/divis1969/android_frameworks_av/commit/6380aabd56953e4a3805ff1de2e4f9762aedd4c2

But you still need some patches to fix HDR, camera options, camera freezing and netd bootloop.

micheleberrettis1, 2017-12-29:
Thank you, achieved to build...

closed 2017-12-29 by micheleberrettis1