Moyster/nougat_device_meizu_m2note #66: band LTE nort america

javijdd, 2018-01-04:
hello moyster this would be very interesting to build some kind of tools to unlock the LTE band for North America and searched but there is little progress on the subject you have created a super rom and created that you can do that work there are also many people waiting for that day. greetings moyter.

Moyster, 2018-03-04:
Answering a bit late but still, Bands can't be added as far as I know, as there's some licensing and stuff behind the 4G bands or something (and thus gets locked when devices aren't meant for release in this part of the world), some modems might have these bands enabled but you'd need to find a similar device (with a really really close matching modem/modem fw) and do trial and error.  
Knowing the mediatek modem sources are obtainable online, one with enough knowledge could rebuild a modem for a mediatek device AND enable additional bands, but even Fire855 told me he couldn't get a self-compiled modem to work properly.  
So, it's pretty much just like VoLTE support (which mt6753 does, but not most of the mt6753 device : k4 note / m2 note, etc..), Mediatek/ODMs are limiting their hardware functionalities to create artificial product ranges (low / mid / high end devices, as long as VoLTE is not a "common thing", it's a "premium thing" to keep selling flagships).  

Welcome to capitalism 👎

closed 2018-03-04 by Moyster

javijdd, 2018-09-28:
hello I know that super late but I saw this model mtk has all the band open a person with a lot of knowledge can help and thanks for answering. greeting.    Modelo umidigi one pro  Ebay.com