Moyster/nougat_device_meizu_m2note #67: Offline charging animation is reverted

aftersilence, 2018-01-11:
The image of offline charging animation is reverted, see a pic attached
[image]
.

Moyster, 2018-03-04:
It's not a bug, you're holding the phone in the wrong orientation :p Joke's aside, that's... yeah not really critical so I never cared to "fix it". Will switch to "help wanted" if anyone has a clue how to rotate 180° healthd offline charging mode.

micheleberrettis1, 2018-03-04:
Might be a silly speculation, but Just rotating of 180° the PNGs that the animation uses?

Moyster, 2018-03-08:
Not sure if this would work, the easiest way would be to rely on ro.sf.hwrotation so healthd could tell what the device need, and flip the buffer to have it displayed in the correct orientation. Considering it doesn't break / prevent anything, it's not really top-priority