Moyster/nougat_device_meizu_m2note #72: Build 22 (Estable 6 "M'lady

javijdd, 2018-01-23:
hello this latest version has failures. the rom stops in less than a minute it goes off and turns on again. This problem is repeated several times when the factory reset continues with the same problem. The second thing I noticed was with the app called whastapp if a friend sends you a voice message is activated wl light sensor passes the hand turns off the screen If you remove your hand, they turn on again, it seems like you are receiving a call and the answer is to hold the phone to your face so that the screen turns off. delo demas no note problem but unfortunately I had to return to the original rom. I hope it is resolved soon.

superlex, 2018-02-07:
Build 22 is very unstable, downgrade to 21 or wait for build 23.
A lot of people is experiencing random reboots and other bugs:
https://forum.xda-developers.com/meizu-m2-note/development/rom-lineageos-14-1-meizu-m2-note-t3692279/page22

Moyster, 2018-03-04:
renamed title so it provides a quick solution for build 22 instabilities.

superlex, 2018-05-26:
Hi,
thank you @Moyster for the new build :)
Build 23 is more stable than 22.

superlex, 2018-06-10:
@dariusworks 
Wait for build 24,  build 23 is more stable than 22 but it's still buggy and sometimes there are system or app crashes.

dariusworks, 2018-06-10:
I'm still on build 22. Use 21 then? @superlex 
Also Thanks!

superlex, 2018-06-10:
According to @Moyster 

> AVC decoder made the whole rom instable since the January '18 security bulletin

(see issue #73 ) so you should use the rom with previous security bulletin (build 20 or 21). However if you are on build 22 maybe the better thing is update to build 23 and wait for 24 one.

dariusworks, 2018-06-10:
I just installed build 21... I'll maybe give 23 a try if 21 isn't that good or something... 
Thanks for help though!

dariusworks, 2018-06-10:
Also, is it me or https://download.lineageos.org/extras 14.1 root zip doesn't work? It worked on older builds...

superlex, 2018-06-10:
You're welcome :)
As regards root, try Magisk 14 as suggested here:
https://forum.xda-developers.com/showpost.php?p=76759228&postcount=333

dariusworks, 2018-07-11:
@superlex Just noticed that sometimes booting fails. Just hold power button and it boots up next time... However this time looks like it's just won't boot, even if I didn't do any sort of changes to system stuff... Trying to boot it up for quite a while now...

superlex, 2018-07-13:
@dariusworks 
You can try to boot to recovery and install LineageOS again. If you can backup your data first, remember that twrp have mtp support.
However, now I'm using build 21: security patches are old but this is very stable, on the contrary both build 22 and build 23 have too many crashes and system reboots.