Moyster/nougat_device_meizu_m2note #76: can't boot after installed rom

SivanLiu, 2018-03-17:
I have build successfully, but it didn't reboot after installed.

micheleberrettis1, 2018-03-17:
Patched Netd? Audioflinger?

SivanLiu, 2018-03-20:
I followed the guide building the rom, not include audioflinger.

micheleberrettis1, 2018-03-21:
You need to patch the sources before u can boot... You need to patch system/netd, frameworks/av/services/audioflinger and frameworks/av/services/camera

SivanLiu, 2018-03-27:
I only modify followed the advice:
https://github.com/Moyster/nougat_device_meizu_m2note/issues/22

zty-zty, 2018-05-23:
i am in  the same trouble,can not boot

Moyster, 2018-05-23:
I actually am too limited in time to do anything, I'll be back at the end of the month.

If you want to help and show support you can go to this twitter page ( https://twitter.com/ARCHOS ) and ask them :  
"where are the Archos Oxygen 70 kernel sources ? Why are you leeching the open-source community ? Why block @oysterized ?"  

If Archos unblocks me, I'll release a fully working patchset to build / boot Nougat ;)

SivanLiu, 2018-05-30:
@Moyster Thanks for your reply! I'll try it do my best.