Moyster/nougat_device_meizu_m2note #81: The touchscreen does not work

den67rus, 2018-07-09:
Not on one of the assemblies does not work touchscreen.

I think it has something to do with replacing the display unit and the touchscreen with the non-original one.

Moyster, 2018-07-09:
do you have the reference for the new parts ? maybe it's just a matter of enabling the correct drivers in the kernel sources, was it working on flyme/stock rom ?

den67rus, 2018-07-09:
On flyme / stock everything works without problems.
I bought a long time ago touchscreen, it is no longer available.

den67rus, 2019-12-11:
@Moyster the touchscreen works without problems in the 13 version of LineageOS assembled by you.
https://forum.xda-developers.com/meizu-mx/orig-development/rom-cm-13-0-meizu-m2-note-t3525030 

The driver in the Assembly added to the core or initrd?